mr-manuel/venus-os_dbus-multiplus-emulator #1: Incorrect calculation of PV Energy

okuegow, 2023-07-27:
Hi Manuel,

I am using your Multiplus Emulator in an environment where I am planning to install a rather large Battery soon.
Really appreciate your script :-)

In order to correctly size the Battery I am first collecting data before buying more equipment.
This is done with 2 EM540 Energy Meters (one for PV Inverter and one for Grid) and a GX device only.
No Multiplus there yet.

The VRM and the GX console are showing correct values of Power Generated and consumed at the moment.
But it seems the Energy amounts it calculates and sends to VRM are exactly DOUBLE the correct amount.
I am getting the correct readings from the Energy Meters and the PV Inverter itself.

Since everything is shown correctly and the only error is coming from the calculation of the energy amounts (PV), I am assuming that there could an error in your emulation script.

If it helps I can share my VRM installation with you.

thanks and best regards,
Olli

mr-manuel, 2023-08-12:
Can you post some values and screenshots? What do you see in the `dbus-spy`?

okuegow, 2023-09-12:
sorry to have wasted you time, problem was on our side.
After a clean re-install of the venus-system everything is being reported just fine.
Can't say for sure what the issue was, but since we had tried 4 different grid&pv meters on this install and replaced them with each other, it could be that for some reason something was being counted double.
all good now.  thanks again for your software.

closed 2023-09-12 by mr-manuel