mr-manuel/venus-os_dbus-multiplus-emulator #2: second Multiplus?

Starter4711, 2024-02-25:
Hello,
I have 3P ESS with MP2 5k and Gobel Packs. 
A second 3P ESS wirh MO2 5k just to additionally load the Batteries from a second Source. 

Would it be possible to get the second MP into the first VRM with your plugin?

First I just want show the VRM, there is a Charger on the battery. 
Second the Cerbo should reduce the Charging Limits of first MP withe the value of the second MP. 

If there are not two MP in one VRM system, maybe a virtual MPPT?

Thanks

mr-manuel, 2024-02-27:
Unfortunately I don't know how it looks like, when there are two or three MultiPlus or Quattro connected to one Venus OS device.

I don't think to alter the system values, because it can lead to system instability, since the system think it's loading with 40 amps, but in real it's only loading with 10 amps.

Starter4711, 2024-02-27:
Thanks for your response. 

It would  just be an additional Info to VRM like a MPPT , the BMS Shows up the Limits for the MP2, that Will Not be a conflict.

mr-manuel, 2024-02-27:
Are you sure? I'm not. Why don't you just add the other MultuPlus to a separate instance of the VRM portal?

A MPPT is a additional device, a MultiPlus is the hearth of the system, which connects DC with AC and vice versa. The MutliPlus therefore also controlls how much current flows in which direction, which in the worst case, could also cause damage.

Sorry, but I won't do this.

closed 2024-02-27 by mr-manuel