mr-manuel/venus-os_dbus-multiplus-emulator #4: Multiple AC Inputs - How to Modify to Show all

mar0997, 2024-05-30:
Thank you for putting this together.  I have a Victron VM-3P75CT energy meter with 2 AC inputs on my boat.  Please note that I have the device currently configured as a 3 phase device while I am actually monitoring 2 single phase input lines.  Victron is planning a future firmware update to handle multiple single phase lines, but it not yet released.  This emulator allows me to properly show an AC input on VRM and on the html5 app on my MFD, which is something that I have been struggling to figure out.  I have a couple of issues that I am hoping you can help me understand.

On VRM the grid input power is properly reflected.  I can see both of my inputs and the total power is correctly the summation of the two inputs.

On VRM the for AC loads I can only see the consumption from one of my lines (L2) while L1 shows 0.  I know from looking at my energy meter that L1 consumption is not 0 and is the same as the grid value.  Can you help point me how I can correct this?

On my MFD app the shore power is only showing the current for L1 and not L2.  And for AC loads I am also only seeing the value for L1 and nothing for L2.  Can you point me to where I can make the shore power current the total of L1 and L2 (same as on VRM grid input)?  And for AC loads how I can also have the summation?

I would be glad to make a donation - hoping to get it working properly :-)
[image]
[image]

mr-manuel, 2024-05-31:
You solved it?

mar0997, 2024-05-31:
Thanks for the quick reply.  I included a very simple wiring diagram.  I know that the energy meter is installed and functioning correctly as I can see the L1 and L2 values in remote console (included screenshot of that).

I discovered that the MFD app is not froze - I needed to reboot the Cerbo and when I did that it is now properly displaying L1.  I still do not see any readings for L2.

I included a couple of Dbus-spy screenshots that show the energy meter and the multibus emulator.  I can see L1 and L2 in in the energy meter Dbus.  But I only see L1 as having values in the multibus emulator.

I looked at the multibus emulator code and I see a couple of lines that talk about "NumberofACInputs".  I tried changing these to "2" and also changing "ACInput2" from "0" to "1", but this did not work.  Am I on the right track?  Or are there other changes that I should be trying?

[image]
[Simple Energy Meter Schematic.pdf](https://github.com/user-attachments/files/15508256/Simple.Energy.Meter.Schematic.pdf)
[image]
[image]

mr-manuel, 2024-05-31:
A MultiPlus has only one input. If you have AC-In 2 then you have a Quattro device. In a real constellation you would need one MultiPlus/Quattro per phase linked together via VE.Config. unfortunately I don't know yet how this looks like in dbus-spy. I have to look if I find something.

mar0997, 2024-05-31:
Ok that makes perfect sense and I should have remembered that the Multiplus has a single input.  One other thought - what if I set the emulator up as 3 phase?  If I changed line 839 "Number of Phases" from "1" to "3" then I wonder if the emulator would pull in L2 (L3 would be 0).  I had to leave my boat for the night or I could have quickly tried that.

mr-manuel, 2024-05-31:
Have you internet access at your boat? If yes, you can use tailscale to access via SSH. [venus-os_TailscaleGX](https://github.com/mr-manuel/venus-os_TailscaleGX)

mar0997, 2024-06-01:
I don't have tailscale yet but am going to download it and get it set up.  That's a really good option.

I went to my boat tonight and the change to 3 phase didn't work.

I did discover that when I changed the inverter phase from "L1" to "L2" that the my MFD app then displayed the readings for my second power input (L2).  So that did make a difference that surprised me.

I tried a couple of other things but didn't have any luck.  If you have any other tip I can always give them a try . It should be easier for me once I install and figure out how to use tailscale.

mr-manuel, 2024-06-01:
You basically need to have a MultiPlus per phase, therefore in this case a MultiPlus emulator for L1 and one for L2. The most clear it would probably be, if you name one `dbus-multiplus-emulator -L1` and the other `dbus-multiplus-emulator -L2`. Then you have to correct all paths, to make it work.

mar0997, 2024-06-01:
Thanks for the tailscale tip.  That is very helpful!  It took me a few minutes to get it working but once I did it worked great.

Adding the 2nd multiplus instance tested my Linux skills but I finally got it.  And it works!  For the most part.  The power readings between input and output looks to be slightly off so I will need to investigate that further.  And I'm not at my boat so I can only go by what I am seeing on VRM and the gui but I do believe it is working better.  To test it I remotely turned on my water heater (which is on L1) and my HVAC (which is in L2).

[image]

[image]

[image]

mr-manuel, 2024-06-01:
All solved now? :-)

mar0997, 2024-06-02:
I thought I had made good progress earlier today but then I ran into similar problems to what I was having for the past couple of days.

One thing I did find is that I must not have successfully added the second multiplied emulator.  I discovered those by noting that the VRM device list only contains the original multiplied emulator and not the new one that I created.  I did have some trouble installing the new one today so I think tomorrow I will start over and uninstall that and reinstall it.  I took the commands that you had provided and added an "L2" to the directory name:

For example I changed the copy file command to:
cp -R /tmp/venus-os_dbus-multiplus-emulator-master/dbus-multiplus-emulatorL2/ /data/etc/

When I did this I had some trouble but did get the files copied.  I then followed the same logic for install.  Is that the correct way to do it?

mr-manuel, 2024-06-02:
Today I got the data from a three phase system with a MultiPlus on each phase. I worked the whole day to fix all and make the emulator one, two and three phase compatible. Install the last version and see, if it works. Only once instance is needed.

mar0997, 2024-06-02:
Thanks for all of the work you put in on this!  I updated the version to the latest this morning and have been doing some troubleshooting.  The grid values are correct.  I am still having trouble with the AC loads.  Here is what I have found:

I updated the user values as shown in the photo.  Changed grid frequency, uncommented voltage for American grid, added DC and AC Dbus services and uncommented phases to be L1 and L2. In the photo timestamped 11:33 you can see that the AC input is correct.  The AC loads not - they jump around slightly but are always near 0.

When I change the phase combination to L1 then I get correct AC load values for L2 but not L1 ( always near 0).  The AC grid input values are correct for both L1 and L2.  This is in the photo timestamped 11:49.

When I change the phase combination to L2 then I get correct AC load values for L1 but not L2 ( always near 0).  The AC grid input values are correct for both L1 and L2. This is in the photo timestamped 11:56.

When I comment out all of the phase combination then the AC grid input values go away (of course).  In this case I get correct AC load values for both L1 and L2.  This is in the photo timestamped 12:12.

So I do feel like we are getting closer.  I haven't had a chance to look at the code further yet but there seems to be some type of interaction when selecting multiple phases that are interfering with the AC load values.

[image]

[image]

[image]

[image]

[image]

mr-manuel, 2024-06-02:
Can you make a wiring diagram of your whole setup with where which device/sensor is located with the different voltages? 120 VAC / 12 VDC

What is converting the power from DC to AC?

Are you using a SmartShunt to fetch the power that comes from the battery and MPPT?

mar0997, 2024-06-02:
Apologies for the crude schematic.  I started with the crude schematic that I sent a couple of days ago and expanded from that.  I do have 2 smartshunts - one for my 2 starboard batteries and one for my 2 port batteries. I do have a small 800 VA inverter but it o my feeds some branch circuits on the L2 input.

So with my particular setup the DC side wouldn't play very much into the overall setup.  What I'm looking at more than anything is getting the grid values properly represented (which they now are).  But in the process of getting those properly represented I have lost the AC loads.  Hope that makes sense?

[image]

mar0997, 2024-06-02:
I see that with the new version the multi is in bulk whereas yesterday it was in pass thru.  If it was in pass thru then the battery connection wouldn't matter right?  I tried to turn it to "OFF" mode but that doesn't work.  Is that something that would help?

mar0997, 2024-06-02:
Ok so I had a discovery.  I went to the boat to pick something up and took a look at my MFD.  And the app readouts are showing that this working as expected.  So it's working fine on the MFD app.  And I also see that it is working on remote console.  See enclosed for photos of both of those.  Yet on VRM and gui (beta) it is not.  So strange.

[image]

[image]

mr-manuel, 2024-07-02:
The state of the emulated multiplus can be ignored, since you have none the state could be whatever. Also other report some issues regarding the load, but currently I have very limited time to look into it.

mr-manuel, 2024-10-11:
This should be solved with the next version. In the current version there are some thinking issues.

closed 2024-10-11 by mr-manuel