mr-manuel/venus-os_dbus-multiplus-emulator #5: Critical Load showing leading to wrong AC Loads

mhin76, 2024-06-05:
great service! I just installed it. I discovered that it shows a critical load, not sure where that number is coming from. That value is added to AC Loads showing incorrect AC Load and Grid Load. Is there a config I am missing?
[image]

mhin76, 2024-06-05:
Update: I added all three phases on the config file, showing now all three:
[image]
my idea first was to not select any phases, but phase 1 (L1) stays with latest value. next idea was now I only select phase 3, as that one is 0. but phase 1 (L1) is then sill listed. So only possibility currently I would have is to wait until L1 is 0, then take out all phases. should be possible once night approaches.

mhin76, 2024-06-05:
and last update: when selecting L3 (which is 0, because no inverter is attached to that phase) it just takes some time to have the critical load reduced to 0 as well! so that approach works for me.
[image]

I will close this issue. hope it helps others that have similar challenge.

mhin76, 2024-06-05:
sorry, for back and forth, just see when selecting L3 the sum is not correct. there is usage from Grid, but AC Loads does not show anything. when selectin L1, then L3 AC load is correct, but critical load is then again messing up the AC load.

closed 2024-06-05 by mhin76

mr-manuel, 2024-06-05:
I'm not using it anymore, therefore I have no feedback for you. Can you post your configuration, if someone else has the same issue?

mhin76, 2024-06-05:
I still have only on VRM a load on L2, even critical load is all 0 on remote console page. I will play a bit more around this evening when there is no inverter power. But I like your code a lot! thanks for sharing!
only config change I did was setting the phase to ="L4". more update later on.

mhin76, 2024-06-07:
I found the solution. all working fine for me now. I set the phase to L4 but also needed to turn off the critical load on the simulated ESS:
[image]

all figures are correct and all data seem to be correct:
[image]