mr-manuel/venus-os_dbus-multiplus-emulator #6: Use MultiPlus Control on VRM

mhin76, 2024-06-20:
Hi Manuel
great driver, works seemlessly with the dbus-mqtt-battery one. 
Not an issue, but rather a new requirement.
As I am controlling the battery charging via nodered (daly BMS) I wonder if there is any chance to get the input from this emulated multiplus on VRM back e.g. to e.g. MQTT. E.g. controls like battery charge on/off, min SOC %, max current limit.
Kind regards,
Martin

mr-manuel, 2024-06-20:
Sorry, I can't completely understand what you need. Since the MultiPlus is emulated you cannot change the On/Off/Invert only/Charge only switch, since it has no action.

mhin76, 2024-06-20:
It would be great if the control input on VRM about battery is somehow getting back to the emulated driver and maybe stored on MQTT, I am using NodeRed and the dbus-mqtt-battery to steer charging/uncharging.
[image]

not sure if VRM is sending anything back to your multiplus emulator.

mhin76, 2024-06-20:
update: I think those settings I should get out from the dedicated Venus OS netatmo nodes :-) will keep it updated here.

mr-manuel, 2024-06-20:
Are you using the latest version of this driver? I made some tests and if you change the settings in VRM, it changes also on Venus OS.

[image]

mhin76, 2024-06-20:
nice, with gif :-) very convenient!
that is aligned, the changes on the vrm control and the venus OS. what I did not manage yet is to get the setting out via NodeRed venus nodes. I can select the MultiPlus (emulated) but the values to get out are very limited:
[image]

but it is low priority! as I can also steer it via iPhone using Homebridge switches.

mr-manuel, 2024-06-20:
You are using the wrong node. This has nothing to do with ESS. Use the `Ve.Bus System` node.

mhin76, 2024-06-20:
Thank you so much Manuel for you support. That was it!! Works 😀💪🏼

mr-manuel, 2024-06-20:
Feel free to buy me a beer on this hot days 😎

closed 2024-06-20 by mr-manuel

mhin76, 2024-06-21:
@mr-manuel happy to do so, just via donation paypal charges 10%! :-o if you send my your IBAN on buymeabeer@hinz.ch I am happy to transfer there without charges :-)

mr-manuel, 2024-06-21:
Not via family & friends, there is no fee: https://paypal.me/mrmanuel3

mhin76, 2024-06-21:
not for me from Switzerland, they want even over 30% of fee :-( last time my donation was also via IBAN to another home automation guy. 
[image]

mr-manuel, 2024-06-21:
Then don't worry :)

mhin76, 2024-06-21:
ok, would have loved to buy you some beers. :-)

mhin76, 2024-06-21:
der Wille war da, via Donation, da kostet es nur 10% Gebühr. Versuche es später nochmals. Weiss nicht, was falsch geht.
[image]

mr-manuel, 2024-06-21:
Very odd. Thanks again