mr-manuel/venus-os_dbus-multiplus-emulator #7: AC load includes the GridMeter value as Output

mhin76, 2024-07-01:
Hi Manuel. I have installed your multiplus emulator. I have latest venus OS installed on my RP3 and an energy meter connected via USB to venus OS directly. Set up as Gridmeter on all 3 phases:
[image]

The grid values show correctly.

However when I select any phase on the emulator config I noticed that the AC loads on the selected phase gets way too high. seems that the grid amount is added on top again.
here the setting with just L1:
[image]

In the log I see that the energy meter is recognized as well by the emulator:
[image]

and it shows under the emulator the power as AC input but also AC output. But there is no output. and exactly that output is added I guess to the AC loads.
[image]

what do I miss?

mr-manuel, 2024-07-02:
Sounds like a bug in the code, since also others describe a similar behaviour: https://github.com/mr-manuel/venus-os_dbus-mqtt-pv/issues/32

Currently I'm very busy and have no time to look at it. Will take a few weeks/months.

mhin76, 2024-07-02:
@mr-manuel I am reengineering and debugging your code. Very nicely written!! And wow, complex requirements. I found out so far that AC out is only for critical loads. maybe with my seput also AC in should only be what goes to battery.
I will let you know if I find something. Thanks for the link above!

mr-manuel, 2024-07-02:
> AC out is only for critical loads

That's true. Maybe this is the problem and the output should always be `0`, but it's only for some cases.

mhin76, 2024-07-02:
@mr-manuel I got it working, I think. In my set-up the AC In should only reflect the power to and from the BMS. Venus OS is then calculating automatically the power from inverter and energy meter. I also set the BMS value (hardcoded) only to that phase where the BMS is connected to. And I set all AC Out to none.

Strange is that the Inveter shows in VRM as off and does not reflect charging/discharging (remote control shows on) and I have not seen yet the charged Wh amount, discharge however shows up. Code looks fine. however I am running the latest Venus OS Beta. Will install latest official release and further test.

mhin76, 2024-07-03:
@mr-manuel  got it working with official venus OS, it shows charging and discharging on the multiplus icon in VRM.
only thing not yet working is the capturing of the battery kWh for charge and discharge. It worked yesterday for 1h. have to find out what I changed on the code.
In general: for me it makes sense that only the battery should be considered for AC in (and having AC out 0 as no critical load should be simulated), because MultiPlus whole purpose is to get and provide power out of the battery only.
I will keep this thread open, until I have solved all issues and will drop you my changes (well indicating what I changed).

mr-manuel, 2024-07-03:
Yes, you are right. The battery should be considered only for AC In.

For the capturing of the kWh of the battery, the battery itself needs to fill the energy forward field.

mhin76, 2024-07-03:
yes, I found that part of the code. 
```
/Energy/InverterToAcOut
...
/Energy/OutToInverter
```
and I like your approach of aggregating it and based on my testing should also be just fine. Will debug later :) need to work now.

mr-manuel, 2024-07-03:
From my tests I did a while back ago, the MultiPlus emulator was not able to provide the kWh from the battery to the VRM portal. You need the [battery device/service](https://github.com/victronenergy/venus/wiki/dbus#battery) to do that. How did you connect the battery?

mhin76, 2024-07-04:
@mr-manuel  works like a charm
[image]

somehow in my set-up the battery was connected but did not provide any current, therefore I had that division by 0 and the program never submitted any Wh. Now it works.
I changed the following area:

1. Excluded AC out (on my code just for one phase and just the power, when I excluded it all, VRM did not like that)
2. Only set AC in to battery power, charge, ...
3. Defaulted my battery voltage to 24 (to prevent division by 0, I tried some kind of if statement, to default it when current = 0 but Python is not my friend... seems we don't like each other :-))

mhin76, 2024-07-04:
just the charging is not showing up. very strange. I see the charging value building up:
```
2024-07-04 11:28:59.686483500 INFO:root:--> data_watt_hours(): {"time_creation": 1720092539, "dc": {"charging": 827, "discharging": 0}, "count": 1}
2024-07-04 11:28:59.712955500 INFO:root:--> charge/discharge: {"dc": {"charging": 0.418, "discharging": 1.484}}
```
and the code submitting it to:
```
self._dbusservice["/Energy/OutToInverter"] = (
            json_data["dc"]["charging"]
            if "dc" in json_data and "charging" in json_data["dc"]
            else 0
        )
```
but nothing in VRM. Will need to check if that is the correct path.

mhin76, 2024-07-04:
@mr-manuel I got it working!!!! 
[image]

finally. don't ask me why, but you have to set the charging amount to the following path:
```
self._dbusservice["/Energy/AcIn2ToInverter"] = (
            json_data["dc"]["charging"]
            if "dc" in json_data and "charging" in json_data["dc"]
            else 0
        )
```
around line 744 in dbus-multiplus-emulator.py. 

I will close this item here.

closed 2024-07-04 by mhin76

mr-manuel, 2024-07-04:
Can you add your script here as attachment so I can compare it?

mhin76, 2024-07-05:
[dbus-multiplus-emulator20240704.2.txt](https://github.com/user-attachments/files/16106842/dbus-multiplus-emulator20240704.2.txt)

yes of course. here we go. but I have not changed it for all phases and on the ACOut I tricked it by asking if "skip" is in the selected phases ;-)