mr-manuel/venus-os_dbus-multiplus-emulator #8: Clarify requirements to use this

msigman, 2024-08-21:
Hello! This project looks great, and may be exactly what I'm looking for.

I have a CerboGX-based system with Victron Phoenix charger and Lynx BMS and wish to have the Cerbo display the charging tile correctly, as if the Phoenix were a Multiplus.

Can you clarify what equipment is necessary in order to use this clever driver?

Thanks

mr-manuel, 2024-08-24:
You need a battery and grid meter to use this, else you have to modify the code.

https://github.com/mr-manuel/venus-os_dbus-multiplus-emulator/blob/8c6d1d2c2d32e288e0684ed14a59cb176bc151df/dbus-multiplus-emulator/dbus-multiplus-emulator.py#L37-L43

If you don't have one of them you can use this:

[venus-os_dbus-mqtt-grid](https://github.com/mr-manuel/venus-os_dbus-mqtt-grid)

[venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery)

msigman, 2024-08-24:
Perfect, thanks very much. Any reason to think this wouldn’t work with gui2?

mr-manuel, 2024-08-24:
It it works with GUIv1 it should also work woth GUIv2

closed 2024-08-25 by mr-manuel